arduino/ArduinoCore-arc32 #551: Serial print not properly works on 2.1.0 rc1

jimaobian, 2017-05-18:
Try "Accelerometer" example in "CurieIMU" library.
It does not get the proper result.

[image]

yashaswini-hanji, 2017-05-18:
Please note that, this is a pre-release for our internal testing. Respective ble_core.bin and quark.bin will be manually placed in Arduino15/packages/Intel/tools/arduino101load/2.0.0/firmwares/2.1.0 during test. Kindly, do not use this tag till we have a official release. Currently we are testing the new features and will have a official release in few days. Thanks!

jimaobian, 2017-05-19:
Hi, take it easy. I know this is a pre-release version. I just played with it as the new version owns more RAM. And I found this issue. Hope to be helpful.

BTW, I'm using the ble_core.bin and quark.bin here: https://github.com/facchinm/arduino101load/tree/master/firmwares/2.1.0