arduino/ArduinoCore-arc32 #555: Enable debug interface on Serail1

hgflas, 2017-05-28:
Hello together,

I would like to enable the debug interface. I have a ftdi USB to TTL adapter connected to rx and tx and the adapter connected to a usb port.

After that I have executed all the steps described under "Enable debug interface on Serail1"

How do I get the log file now? If I initialize Serial1 and open the monitor at the appropriate com port nothing happens.

My code looks like this:
```
void setup() {
  Serial.begin(9600);    // initialize serial communication
  Serial1.begin(115200);
  log_init();
}
```

bigdinotech, 2017-06-02:
@hbisby can you describe how you are connecting to your host PC?
how is the FTDI cable wired and what settings are you using for your serial terminal application

sgbihu, 2017-06-13:
@hgflas ,

To output debug info to the Serial1, you can connect the USB-UART's RX to digital Pin1, GND to GND. And then try to output something to Serial1 to see what output in your console, like minicom in Linux. 
`void setup() {
  Serial.begin(9600);    // initialize serial communication
  Serial1.begin(115200);
  // log_init();  // This line is not necessary.
  Serial1.println("Test for Serial1"); 
}`