arduino/ArduinoCore-arc32 #557: SoftwareSerial does not work properly

karthikeyanprakash, 2017-06-22:
I have tried using Pin 2 and # as software serial to communicate to ESP8266 (AT commands). Unfortunately, the code works properly on UNO but not on 101. I just edited the basic software serial example provided with Lib.

bigdinotech, 2017-06-22:
Hi @karthikeyanprakash 
what baud rate/s are you using?

karthikeyanprakash, 2017-06-22:
@bigdinotech  I am using 9600. The issue is with some of the W10 devices. While it works perfectly with other.

Gerriko, 2017-08-12:
I'm having problems too. The software serial RX receives data but the TX pin does not work. I have tried the example given in CurieSoftwareSerial and TX does not work. I have tried digital pins 2 - 12, nothing. I have tried baud rate 9600 and 38400. I have tried to write just 1 byte to TX pin and receive nothing on the other end - I note that in the library only a write method exists so not sure how the println would work.

bigdinotech, 2017-08-14:
Hi @Gerriko ,
Can you try https://github.com/01org/corelibs-arduino101/commit/c794c47abf9864af34b768f46f0792991f12a649

Gerriko, 2017-08-15:
Yes, I can try it. But before I start messing with my Curie SoftwareSerial.cpp file please provide some brief instruction on how to get the required amended version of cpp file or am I to cut and paste fix myself. Not clear on GitHub page as only see a unified view of change.

closed 2017-08-16 by bigdinotech