arduino/ArduinoCore-arc32 #565: Arduino 101 / Arduino 101 stuck after uploading code

tamirmha, 2017-08-16:
have an arduino 101 and after i added to my project an EEPROM Get/Put its worked for 1 minute and after this the device stucked and didn't response anymore and the computer doesnt recognize it any more ( i disconnected from the usb, tried the master reset, tried to downgrade for older version the ide and the drivers).
i tried before to write the only to the eeprom and its worked fine
i have win7 64bit and the latest version of the ide and the intel curie drivers)
the device is like a brick now
I attached the code.
Hope if anyone could help
[BLE_Arduino1.zip](https://github.com/01org/corelibs-arduino101/files/1228655/BLE_Arduino1.zip)

bigdinotech, 2017-08-16:
Hi @tamirmha ,
Can you try to reflash the firmware.
Tools-> Burn Bootloader
Make sure that the programmer selected is Arduino/Genuino 101 Firmware Updater"

tamirmha, 2017-08-16:
Hello Bigdinotech,
the computer doesn't recognize the device any more and the port didn't appear also.
i tried this and its didnt work. its give the following message:
"Starting download script...
Flashing is taking longer than expected
Try pressing MASTER_RESET button
ERROR: Timed out waiting for Arduino 101 on COM7
Error while burning bootloader."

IeiuniumLux, 2020-03-15:
Will this ever be fixed?

PaulStoffregen, 2020-03-19:
Who do you imagine will work on this problem?

Intel discontinued the chip.  Arduino no longer sells this board.  Intel reassigned or laid off all the developers who worked on the software.

Seems extremely unlikely any lingering issues with Arduino 101 or Intel Curie will ever be fixed.

AmanuelEphrem, 2020-05-29:
If you are having uploading issues, I have a fix that helped me. Upload your code to your board and press the master reset every 3-4 seconds. I know it is jankey, but it worked for me :)

Rafmigz, 2020-11-11:
Hi! I also have an Arduino 101 when I upload a code, the Arduino will have a sound of disconnecting USB and connecting sound, until an error will appear: "ERROR: Timed out waiting for Arduino 101 on COM6" is there a fix on this? Thanks!

eriknyquist, 2020-11-13:
@Rafmigz  @IeiuniumLux @tamirmha 

My apologies for never responding to this thread-- I had these emails tucked away in an old gmail filter because, well, Intel laid us all off like Paul said :P

Everything @PaulStoffregen said is correct, and because of this I think it is very unlikely that any further development or bug fixing will happen for this repository. This is only speculation of course, since I don't work at Intel anymore, but it seems like a safe bet given the lack of recent activity on this repository.

Thanks, and good luck.
Erik

Rafmigz, 2020-11-16:
Hi! I was really persistent to look for ways on how to fix the Arduino/Genuino 101 Timeout Error, I searched different threads and read a lot of suggestions on how to fix the error, but still all failed, I know some of the 101 users already bricked their 101 or other just left it behind. 

We are all aware that the Arduino 101 is already discontinued and even support is already stopped, Hence this little but powerful microcontroller is still out there and probably used by the some. Anyhow, to be straight to the point I finally I discovered the fix for the error.

Notice that when you insert your 101 to the USB plug it will read and install the device, checking it in the device manager will show the Arduino/Genuino 101 in "Other Devices" and a new COMPort (COMX) is added, then you install the Arduino IDE (regardless of the version), then install the Arduino 101 Board thru "Board Manager", the manager will install the Arduino 101 but it will not notify you that the driver for Arduino 101 is not installed due to error. So in short we are assuming that the driver for Arduino 101 is successfully installed. This is where the error lies. The software driver "dpinst-amd64" is not digitally signed and Windows is blocking to install unsigned drivers. 

So what's the fix? 

The fix here is still to install the "dpinst-amd64" since it is the driver needed to recognize your 101, the trick here is you need to install it in other way, not in your regular PC operation. You need to access the "disable signature enforcement" under the startup options (not safe mode).  This startup mode will enable you to install unsigned drivers (I trust this Intel Curie driver since it was downloaded with the board). After installation restart your PC to its regular startup and plug-in again your Arduino/Genuino 101, this time the PC will now recognize the microcontroller and you can now upload sketches without the time out error.

I hope this will help. :D I will try to make a video about this fix. :D

Thanks!

Regards,

Raf

Shlumpith, 2020-11-19:
I tried to use the solution Rafmigz was able to find. When I ran the driver setup wizard it was able to install the intel drivers but not the libusb drivers. I still cant upload code to the board yet, as it will still wait on the device. Any ideas on how to fix this?
[image]
^ image of the dipnst-amd64 driver installer failing, I ran the program as an administrator.

Rafmigz, 2020-11-27:
Hmmm, seems another problem, try to do it in other computer, as I tried mine for 4 computers and everything seems to work well with the fix.  

or try to re install the driver again. 

Thanks!

Rafmigz

Syntrice, 2021-01-07:
> Hi! I was really persistent to look for ways on how to fix the Arduino/Genuino 101 Timeout Error, I searched different threads and read a lot of suggestions on how to fix the error, but still all [...]

I can confirm this worked for me. Finally glad to have a fix!

Yasu31, 2021-01-13:
@Rafmigz 's solution worked, thank you! Leaving a step-by-step memo of how I resolved it, for future reference...
1. disconnect the Arduino 101 if it is connected (not sure if this is necessary, but just in case to prevent if Windows decides to automatically install drivers)
1. uninstall the **Intel Curie Boards** from the Arduino Boards Manager
2. restart Windows into the boot options menu. I used the `shutdown.exe /r /o` command from Windows Powershell as Admin (ref: https://www.howtogeek.com/126016/three-ways-to-access-the-windows-8-boot-options-menu/ )
3. In the boot options menu, choose *Troubleshoot* -> *Advanced Options* -> *Restart*
4. After restart, choose *Disable driver signature enforcement*
5. Install **Intel Curie Boards** again, from the Arduino Boards Manager

wengweng675, 2021-03-18:
> @Rafmigz 's solution worked, thank you! Leaving a step-by-step memo of how I resolved it, for future reference...
> 
> 1. disconnect the Arduino 101 if it is connected (not sure if this is necessary, but just in case to prevent if Windows decides to automatically install drivers)
> 2. uninstall the **Intel Curie Boards** from the Arduino Boards Manager
> 3. restart Windows into the boot options menu. I used the `shutdown.exe /r /o` command from Windows Powershell as Admin (ref: https://www.howtogeek.com/126016/three-ways-to-access-the-windows-8-boot-options-menu/ )
> 4. In the boot options menu, choose _Troubleshoot_ -> _Advanced Options_ -> _Restart_
> 5. After restart, choose _Disable driver signature enforcement_
> 6. Install **Intel Curie Boards** again, from the Arduino Boards Manager

how to uninstall the intel curie board

PaulStoffregen, 2021-03-18:
@per1234 - Perhaps the Arduino 101 retired product page should be updated with a link to this github issue?

https://github.com/arduino/Arduino

The "Getting Started" section on the Overview tab seems like an ideal place to mention how Windows users should install the unsigned driver.