arduino/ArduinoCore-arc32 #580: Is Arduino101 abandoned?

vshymanskyy, 2017-12-28:
I am (and the whole Blynk Community is) waiting for a rather simple issue to be fixed: https://github.com/arduino/ArduinoCore-arc32/issues/531.
We're waiting since April. Is Arduino101 abandoned?

eriknyquist, 2017-12-31:
I cannot say what Intel has decided to do with arduino101 (because I do not know), but I can say that I and everyone else who worked on arduino101 has been laid off. We all work elsewhere now.

vshymanskyy, 2018-01-07:
@eriknyquist Thank you for your work, anyway. It was a promising device among makers at least ;)

eriknyquist, 2018-01-08:
it was a fun ride, I'm not complaining :) also, I've seen no announcements that Intel is cancelling Arduino101, it might still live on. Of course someone else has to get up to speed so threads might just be silent for a while (just speculating here, don't take that for an official update....)

foxed-in, 2018-01-08:
Check out the new home (URL) for this repo!

vshymanskyy, 2018-01-08:
@foxed-in could you elaborate?

wldevries, 2018-05-15:
It seems the board is very well supported by the [Zephyr project](http://docs.zephyrproject.org/boards/x86/arduino_101/doc/board.html), a real-time OS by the Linux foundation. It'll probably be more involved to create stuff but the documentations looks good.

I can't really make heads or tails of this repository, because there are no build scripts and it's unclear what runs on the ARC chip and what runs on the x86 chip .. nor any pointers on how the build all the stuff and create the base firmware. Is this repository built on the Zephyr project or something else?

PaulStoffregen, 2018-05-15:
From your comments, I'm guessing you've never used the Arduino software?

To build this repository, you do the same as any other of Arduino's boards.  From the Arduino software, you click Tools > Boards > Boards_Manager.  Then from the list of board packages to install, select "Intel Curie (32-bit) Boards".  Wait for a moment while the software downloads and installs this repository and the related tools.

Then click Tools > Boards to select "Arduino/Genuino 101", and click Tools > Ports to select which serial port is for the 101 board you wish to program.  If you don't have a program open, perhaps open any of the examples from the File > Examples menus.  Then click the Upload button in Arduino's toolbar.  The Arduino software will build your code, this code, any libraries your program uses, link it all together and automatically program it onto the board you selected.

It's all meant to be very easy for beginners.  You can say "nor any pointers on how the build all the stuff", but in fact Arduino is pretty much a household name with millions of users and hundreds of tutorials scattered across the web showing absolute beginners how to use the Arduino IDE software.

wldevries, 2018-05-16:
Thanks for the explanation @PaulStoffregen , however I do know my way around the Aruino software. I was just wondering how to build this repository. Browsing through the release notes it seems it is built on Zephyr and from the name of the repository I guess everything here is destined for the Arc core. Since all source code is just packed in a zip in the release it is probably used as a library by the Arduino software to transform and build the .ino?

It would be great if there were some pointers in the form of a readme on how to install a develop version of the board library in the Arduino IDE, what Zephyr version it is linked to and how to flash the 101 with it.

PaulStoffregen, 2018-05-16:
If teenagers all over the world can figure out how to run Arduino, successfully compiling and uploading code to their Arduino boards, I'm pretty sure you can too.

Juanma24-, 2018-06-07:
https://hackaday.com/2017/07/25/the-end-of-arduino-101-intel-leaves-maker-market/

wldevries, 2021-02-26:
@PaulStoffregen no need to be pedantic. I was not asking how to build a sketch, but how to build the firmware for the 101 board.