arduino/ArduinoCore-arc32 #582: Error compiling for board Arduino/Genuino 101.

Raffprta, 2018-01-19:
When I follow the instructions on how to import this project into the Arduino15 directory I get "Error compiling for board Arduino/Genuino 101." after the compilation stage. Are there any details on how to import/modify this project and have it successfully compile using the Arduino IDE?

facchinm, 2018-01-19:
Hi @Raffprta ,
could you paste the **verbose** output of the compilation process?

Raffprta, 2018-01-19:
C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\b3026320\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\b3026320\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries H:\Arduino\libraries -fqbn=Intel:arc32:arduino_101 -ide-version=10802 -build-path C:\Users\b3026320\AppData\Local\Temp\arduino_build_314288 -warnings=all -build-cache C:\Users\b3026320\AppData\Local\Temp\arduino_cache_508660 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduino101load.path=C:\Users\b3026320\AppData\Local\Arduino15\packages\Intel\tools\arduino101load\2.0.1 -prefs=runtime.tools.dfu-util.path=C:\Users\b3026320\AppData\Local\Arduino15\packages\arduino\tools\dfu-util\0.9.0-arduino1 -prefs=runtime.tools.arc-elf32.path=C:\Users\b3026320\AppData\Local\Arduino15\packages\Intel\tools\arc-elf32\1.6.9+1.0.1 -prefs=runtime.tools.openocd.path=C:\Users\b3026320\AppData\Local\Arduino15\packages\Intel\tools\openocd\0.9.0+0.1 -prefs=runtime.tools.flashpack.path=C:\Users\b3026320\AppData\Local\Arduino15\packages\Intel\tools\flashpack\2.0.0 -verbose C:\Users\LocalAdmin\Documents\FuzzySense\ExperimentArduinoProject\Experiment\Experiment.c\Experiment.c.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\b3026320\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\b3026320\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries H:\Arduino\libraries -fqbn=Intel:arc32:arduino_101 -ide-version=10802 -build-path C:\Users\b3026320\AppData\Local\Temp\arduino_build_314288 -warnings=all -build-cache C:\Users\b3026320\AppData\Local\Temp\arduino_cache_508660 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduino101load.path=C:\Users\b3026320\AppData\Local\Arduino15\packages\Intel\tools\arduino101load\2.0.1 -prefs=runtime.tools.dfu-util.path=C:\Users\b3026320\AppData\Local\Arduino15\packages\arduino\tools\dfu-util\0.9.0-arduino1 -prefs=runtime.tools.arc-elf32.path=C:\Users\b3026320\AppData\Local\Arduino15\packages\Intel\tools\arc-elf32\1.6.9+1.0.1 -prefs=runtime.tools.openocd.path=C:\Users\b3026320\AppData\Local\Arduino15\packages\Intel\tools\openocd\0.9.0+0.1 -prefs=runtime.tools.flashpack.path=C:\Users\b3026320\AppData\Local\Arduino15\packages\Intel\tools\flashpack\2.0.0 -verbose C:\Users\LocalAdmin\Documents\FuzzySense\ExperimentArduinoProject\Experiment\Experiment.c\Experiment.c.ino
Using board 'arduino_101' from platform in folder: C:\Users\b3026320\AppData\Local\Arduino15\packages\Intel\hardware\arc32\2.0.2
Using core 'arduino' from platform in folder: C:\Users\b3026320\AppData\Local\Arduino15\packages\Intel\hardware\arc32\2.0.2
Detecting libraries used...
"C:\Users\b3026320\AppData\Local\Arduino15\packages\Intel\tools\arc-elf32\1.6.9+1.0.1/bin/arc-elf32-g++" -c -mcpu=quarkse_em -mlittle-endian -g -Os -Wall -fno-reorder-functions -fno-asynchronous-unwind-tables -fno-omit-frame-pointer -fno-defer-pop -Wno-unused-but-set-variable -Wno-main -ffreestanding -fno-stack-protector -mno-sdata -ffunction-sections -fdata-sections -fsigned-char  -fno-rtti -fno-exceptions -fcheck-new -D__ARDUINO_ARC__ -std=c++11 -DCONFIG_BLUETOOTH_PERIPHERAL -DCONFIG_BLUETOOTH_CENTRAL -DCONFIG_BLUETOOTH_GATT_CLIENT -w -x c++ -E -CC -DF_CPU=32000000L -DARDUINO=10802 -DARDUINO_ARC32_TOOLS -DARDUINO_ARCH_ARC32 -D__CPU_ARC__ -DCLOCK_SPEED=32 -DCONFIG_SOC_GPIO_32 -DCONFIG_SOC_GPIO_AON -DINFRA_MULTI_CPU_SUPPORT -DCFW_MULTI_CPU_SUPPORT -DHAS_SHARED_MEM "-IC:\Users\b3026320\AppData\Local\Arduino15\packages\Intel\hardware\arc32\2.0.2\system/libarc32_arduino101/common" "-IC:\Users\b3026320\AppData\Local\Arduino15\packages\Intel\hardware\arc32\2.0.2\system/libarc32_arduino101/drivers" "-IC:\Users\b3026320\AppData\Local\Arduino15\packages\Intel\hardware\arc32\2.0.2\system/libarc32_arduino101/bootcode" "-IC:\Users\b3026320\AppData\Local\Arduino15\packages\Intel\hardware\arc32\2.0.2\system/libarc32_arduino101/framework/include"  "-IC:\Users\b3026320\AppData\Local\Arduino15\packages\Intel\hardware\arc32\2.0.2\cores\arduino" "-IC:\Users\b3026320\AppData\Local\Arduino15\packages\Intel\hardware\arc32\2.0.2\variants\arduino_101" "C:\Users\b3026320\AppData\Local\Temp\arduino_build_314288\sketch\Experiment.c.ino.cpp" -o "nul"
"C:\Users\b3026320\AppData\Local\Arduino15\packages\Intel\tools\arc-elf32\1.6.9+1.0.1/bin/arc-elf32-g++" -c -mcpu=quarkse_em -mlittle-endian -g -Os -Wall -fno-reorder-functions -fno-asynchronous-unwind-tables -fno-omit-frame-pointer -fno-defer-pop -Wno-unused-but-set-variable -Wno-main -ffreestanding -fno-stack-protector -mno-sdata -ffunction-sections -fdata-sections -fsigned-char  -fno-rtti -fno-exceptions -fcheck-new -D__ARDUINO_ARC__ -std=c++11 -DCONFIG_BLUETOOTH_PERIPHERAL -DCONFIG_BLUETOOTH_CENTRAL -DCONFIG_BLUETOOTH_GATT_CLIENT -w -x c++ -E -CC -DF_CPU=32000000L -DARDUINO=10802 -DARDUINO_ARC32_TOOLS -DARDUINO_ARCH_ARC32 -D__CPU_ARC__ -DCLOCK_SPEED=32 -DCONFIG_SOC_GPIO_32 -DCONFIG_SOC_GPIO_AON -DINFRA_MULTI_CPU_SUPPORT -DCFW_MULTI_CPU_SUPPORT -DHAS_SHARED_MEM "-IC:\Users\b3026320\AppData\Local\Arduino15\packages\Intel\hardware\arc32\2.0.2\system/lib [... cut ...]
Using library LinkedList-master at version 1.2.3 in folder: H:\Arduino\libraries\LinkedList-master 
Using library CurieTimerOne at version 1.0 in folder: C:\Users\b3026320\AppData\Local\Arduino15\packages\Intel\hardware\arc32\2.0.2\libraries\CurieTimerOne 
Using library SPI at version 1.0 in folder: C:\Users\b3026320\AppData\Local\Arduino15\packages\Intel\hardware\arc32\2.0.2\libraries\SPI 
Using library SD at version 1.1.1 in folder: C:\Program Files (x86)\Arduino\libraries\SD 
exit status 1
Error compiling for board Arduino/Genuino 101.

facchinm, 2018-01-19:
The error is about upload, not compilation. You need an updated version of `arduino101load` which has not been shipped yet. You can grab the latest binary from [here](http://downloads.arduino.cc/tools/arduino101load-2.2.2-windows_386.tar.bz2) and extract it in `C:\Users\b3026320\AppData\Local\Arduino15\packages\Intel\tools\arduino101load\2.0.1/` (replacing `arduino101load.exe` binary)

Raffprta, 2018-01-19:
Perfect, that resolves the issue, thank you :)

closed 2018-01-19 by Raffprta

trenskow, 2022-08-28:
If I wanted to download the latest version (2.2.2), but for Mac, then what's the URL? The one you posted four years ago is for Windows.

Luthfidwip1, 2023-05-07:
Arduino: 1.8.19 (Windows 10), Board: "Arduino/Genuino 101"

In file included from C:\Users\Luthfi\Documents\Arduino\libraries\grbl\examples\grblUpload\grblUpload.ino:27:0:

C:\Users\Luthfi\Documents\Arduino\libraries\grbl/grbl.h:29:20: fatal error: avr/io.h: No such file or directory

 #include <avr/io.h>

                    ^

compilation terminated.

exit status 1

Error compiling for board Arduino/Genuino 101.

Does anyone know about this kind of problem?

oqibidipo, 2023-07-09:
> Grbl is a no-compromise, high performance, low cost alternative to parallel-port-based motion control for CNC milling. It will run on a vanilla Arduino (Duemillanove/Uno) **as long as it sports an Atmega 328**.

The [Grbl wiki](https://github.com/gnea/grbl/wiki) lists a few alternatives, but none of them seem to support Arduino 101.