arduino/ArduinoCore-arc32 #596: CurieBLE - BLECommon dosen't existe ..

MRusnak1, 2019-03-28:
"Arduino: 1.8.9 (Windows Store 1.8.21.0) (Windows 10), Board: "Arduino Mega ADK"

In file included from C:\Users\User\Documents\Arduino\libraries\CurieBLE\src/CurieBLE.h:33:0,

                 from C:\Users\User\Documents\Arduino\libraries\CurieBLE\src/BLECharacteristic.h:23,

                 from C:\Users\User\Desktop\26-03-BLE_central.ino\26-03-BLE_central.ino.ino:3:

C:\Users\User\Documents\Arduino\libraries\CurieBLE\src/BLECommon.h:26:54: fatal error: ../src/services/ble_service/ble_protocol.h: No such file or directory

compilation terminated.

exit status 1
Error compiling for board Arduino Mega ADK."

hi guys, need some help ... i can't find a common.cpp on library (CurieBLE) just have a common.h ..
where a i can find a full version of this library .. or how a can change it to put ut work?
thanks

facchinm, 2019-03-29:
Hi @MRusnak1 ,
you need to select `Arduino/Genuino 101` as a target board when using CurieBLE library; right now you are selecting `Mega ADK` :wink:

closed 2019-03-29 by facchinm

MRusnak1, 2019-04-02:
Hi 
Did you know some library whit the same work mode?
I have two boards, but bout of them are mega xD 

Enviado do Correio para Windows 10

De: Martino Facchin
Enviado: 29 de março de 2019 08:11
Para: arduino/ArduinoCore-arc32
Cc: MRusnak1; Mention
Assunto: Re: [arduino/ArduinoCore-arc32] CurieBLE - BLECommon dosen't existe.. (#596)

Hi @MRusnak1 ,
you need to select Arduino/Genuino 101 as a target board when using CurieBLE library; right now you are selecting Mega ADK 😉