arduino/ArduinoCore-arc32 #603: When I try to upload a simple sketch to the Arduino 101, I get the following error on Mac M1 (macOS Monterey)

trenskow, 2022-08-28:
```
/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/trenskow/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/trenskow/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/trenskow/Documents/Arduino/libraries -fqbn=Intel:arc32:arduino_101 -vid-pid=8087_0AB6 -ide-version=10819 -build-path /var/folders/pz/kz0b20y15v7_hpvd7db9067c0000gn/T/arduino_build_193391 -warnings=none -build-cache /var/folders/pz/kz0b20y15v7_hpvd7db9067c0000gn/T/arduino_cache_825497 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduino101load.path=/Users/trenskow/Library/Arduino15/packages/Intel/tools/arduino101load/2.0.1 -prefs=runtime.tools.arduino101load-2.0.1.path=/Users/trenskow/Library/Arduino15/packages/Intel/tools/arduino101load/2.0.1 -prefs=runtime.tools.dfu-util.path=/Users/trenskow/Library/Arduino15/packages/arduino/tools/dfu-util/0.9.0-arduino1 -prefs=runtime.tools.dfu-util-0.9.0-arduino1.path=/Users/trenskow/Library/Arduino15/packages/arduino/tools/dfu-util/0.9.0-arduino1 -prefs=runtime.tools.openocd.path=/Users/trenskow/Library/Arduino15/packages/Intel/tools/openocd/0.9.0+0.1 -prefs=runtime.tools.openocd-0.9.0+0.1.path=/Users/trenskow/Library/Arduino15/packages/Intel/tools/openocd/0.9.0+0.1 -prefs=runtime.tools.arc-elf32.path=/Users/trenskow/Library/Arduino15/packages/Intel/tools/arc-elf32/1.6.9+1.0.1 -prefs=runtime.tools.arc-elf32-1.6.9+1.0.1.path=/Users/trenskow/Library/Arduino15/packages/Intel/tools/arc-elf32/1.6.9+1.0.1 -prefs=runtime.tools.flashpack.path=/Users/trenskow/Library/Arduino15/packages/Intel/tools/flashpack/2.0.0 -prefs=runtime.tools.flashpack-2.0.0.path=/Users/trenskow/Library/Arduino15/packages/Intel/tools/flashpack/2.0.0 -verbose /Users/trenskow/Documents/Arduino/sketch_aug27a/sketch_aug27a.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/trenskow/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/trenskow/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/trenskow/Documents/Arduino/libraries -fqbn=Intel:arc32:arduino_101 -vid-pid=8087_0AB6 -ide-version=10819 -build-path /var/folders/pz/kz0b20y15v7_hpvd7db9067c0000gn/T/arduino_build_193391 -warnings=none -build-cache /var/folders/pz/kz0b20y15v7_hpvd7db9067c0000gn/T/arduino_cache_825497 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduino101load.path=/Users/trenskow/Library/Arduino15/packages/Intel/tools/arduino101load/2.0.1 -prefs=runtime.tools.arduino101load-2.0.1.path=/Users/trenskow/Library/Arduino15/packages/Intel/tools/arduino101load/2.0.1 -prefs=runtime.tools.dfu-util.path=/Users/trenskow/Library/Arduino15/packages/arduino/tools/dfu-util/0.9.0-arduino1 -prefs=runtime.tools.dfu-util-0.9.0-arduino1.path=/Users/trenskow/Library/Arduino15/packages/arduino/tools/dfu-util/0.9.0-arduino1 -prefs=runtime.tools.openocd.path=/Users/trenskow/Library/Arduino15/packages/Intel/tools/openocd/0.9.0+0.1 -prefs=runtime.tools.openocd-0.9.0+0.1.path=/Users/trenskow/Library/Arduino15/packages/Intel/tools/openocd/0.9.0+0.1 -prefs=runtime.tools.arc-elf32.path=/Users/trenskow/Library/Arduino15/packages/Intel/tools/arc-elf32/1.6.9+1.0.1 -prefs=runtime.tools.arc-elf32-1.6.9+1.0.1.path=/Users/trenskow/Library/Arduino15/packages/Intel/tools/arc-elf32/1.6.9+1.0.1 -prefs=runtime.tools.flashpack.path=/Users/trenskow/Library/Arduino15/packages/Intel/tools/flashpack/2.0.0 -prefs=runtime.tools.flashpack-2.0.0.path=/Users/trenskow/Library/Arduino15/packages/Intel/tools/flashpack/2.0.0 -verbose /Users/trenskow/Documents/Arduino/sketch_aug27a/sketch_aug27a.ino
Using board 'arduino_101' from platform in folder: /Users/trenskow/Library/Arduino15/packages/Intel/hardware/arc32/2.0.5
Using core 'arduino' from platform in folder: /Users/trenskow/Library/Arduino15/packages/Intel/hardware/arc32/2.0.5
Detecting libraries used...
/Users/trenskow/Library/Arduino15/packages/Intel/tools/arc-elf32/1.6.9+1.0.1/bin/arc-elf32-g++ -c -mcpu=quarkse_em -mlittle-endian -g -Os -Wall -fno-reorder-functions -fno-asynchronous-unwind-tables -fno-omit-frame-pointer -fno-defer-pop -Wno-unused-but-set-variable -Wno-main -ffreestanding -fno-stack-protector -mno-sdata -ffunction-sections -fdata-sections -fsigned-char -fno-rtti -fno-exceptions -fcheck-new -D__ARDUINO_ARC__ -std=c++11 -DCONFIG_BLUETOOTH_PERIPHERAL -DCONFIG_BLUETOOTH_CENTRAL -DCONFIG_BLUETOOTH_GATT_CLIENT -w -x c++ -E -CC -DF_CPU=32000000L -DARDUINO=10819 -DARDUINO_ARC32_TOOLS -DARDUINO_ARCH_ARC32 -D__CPU_ARC__ -DCLOCK_SPEED=32 -DCONFIG_SOC_GPIO_32 -DCONFIG_SOC_GPIO_AON -DINFRA_MULTI_CPU_SUPPORT -DCFW_MULTI_CPU_SUPPORT -DHAS_SHARED_MEM -I/Users/trenskow/Library/Arduino15/packages/Intel/hardware/arc32/2.0.5/system/libarc32_arduino101/common -I/Users/trenskow/Library/Arduino15/packages/Intel/hardware/arc32/2.0.5/system/libarc32_arduino101/drivers -I/Users/trenskow/Library/Arduino15/packages/Intel/hardware/arc32/2.0.5/system/libarc32_arduino101/bootcode -I/Users/trenskow/Library/Arduino15/packages/Intel/hardware/arc32/2.0.5/system/libarc32_arduino101/framework/include -I/Users/trenskow/Library/Arduino15/packages/Intel/hardware/arc32/2.0.5/cores/arduino -I/Users/trenskow/Library/Arduino15/packages/Intel/hardware/arc32/2.0.5/variants/arduino_101 /var/folders/pz/kz0b20y15v7_hpvd7db9067c0000gn/T/arduino_build_193391/sketch/sketch_aug27a.ino.cpp -o /dev/null
Generating function prototypes...
/Users/trenskow/Library/Arduino15/packages/Intel/tools/arc-elf32/1.6.9+1.0.1/bin/arc-elf32-g++ -c -mcpu=quarkse_em -mlittle-endian -g -Os -Wall -fno-reorder-functions -fno-asynchronous-unwind-tables -fno-omit-frame-pointer -fno-defer-pop -Wno-unused-but-set-variable -Wno-main -ffreestanding -fno-stack-protector -mno-sdata -ffunction-sections -fdata-sections -fsigned-char -fno-rtti -fno-exceptions -fcheck-new -D__ARDUINO_ARC__ -std=c++11 -DCONFIG_BLUETOOTH_PERIPHERAL -DCONFIG_BLUETOOTH_CENTRAL -DCONFIG_BLUETOOTH_GATT_CLIENT -w -x c++ -E -CC -DF_CPU=32000000L -DARDUINO=10819 -DARDUINO_ARC32_TOOLS -DARDUINO_ARCH_ARC32 -D__CPU_ARC__ -DCLOCK_SPEED=32 -DCONFIG_SOC_GPIO_32 -DCONFIG_SOC_GPIO_AON -DINFRA_MULTI_CPU_SUPPORT -DCFW_MULTI_CPU_SUPPORT -DHAS_SHARED_MEM -I/Users/trenskow/Library/Arduino15/packages/Intel/hardware/arc32/2.0.5/system/libarc32_arduino101/common -I/Users/trenskow/Library/Arduino15/packages/Intel/hardware/arc32/2.0.5/system/libarc32_arduino101/drivers -I/Users/trenskow/Library/Arduino15/packages/Intel/hardware/arc32/2.0.5/system/libarc32_arduino101/bootcode -I/Users/trenskow/Library/Arduino15/packages/Intel/hardware/arc32/2.0.5/system/libarc32_arduino101/framework/include -I/Users/trenskow/Library/Arduino15/packages/Intel/hardware/arc32/2.0.5/cores/arduino -I/Users/trenskow/Library/Arduino15/packages/Intel/hardware/arc32/2.0.5/variants/arduino_101 /var/folders/pz/kz0b20y15v7_hpvd7db9067c0000gn/T/arduino_build_193391/sketch/sketch_aug27a.ino.cpp -o /var/folders/pz/kz0b20y15v7_hpvd7db9067c0000gn/T/arduino_build_193391/preproc/ctags_target_for_gcc_minus_e.cpp
/Applications/Arduino.app/Contents/Java/tools-builder/ctags/5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives /var/folders/pz/kz0b20y15v7_hpvd7db9067c0000gn/T/arduino_build_193391/preproc/ctags_target_for_gcc_minus_e.cpp
Compiling sketch...
/Users/trenskow/Library/Arduino15/packages/Intel/tools/arc-elf32/1.6.9+1.0.1/bin/arc-elf32-g++ -c -mcpu=quarkse_em -mlittle-endian -g -Os -Wall -fno-reorder-functions -fno-asynchronous-unwind-tables -fno-omit-frame-pointer -fno-defer-pop -Wno-unused-but-set-variable -Wno-main -ffreestanding -fno-stack-protector -mno-sdata -ffunction-sections -fdata-sections -fsigned-char -MMD -fno-rtti -fno-exceptions -fcheck-new -D__ARDUINO_ARC__ -std=c++11 -DCONFIG_BLUETOOTH_PERIPHERAL -DCONFIG_BLUETOOTH_CENTRAL -DCONFIG_BLUETOOTH_GATT_CLIENT -DF_CPU=32000000L -DARDUINO=10819 -DARDUINO_ARC32_TOOLS -DARDUINO_ARCH_ARC32 -D__CPU_ARC__ -DCLOCK_SPEED=32 -DCONFIG_SOC_GPIO_32 -DCONFIG_SOC_GPIO_AON -DINFRA_MULTI_CPU_SUPPORT -DCFW_MULTI_CPU_SUPPORT -DHAS_SHARED_MEM -I/Users/trensk [... cut ...] uino_101_a1fd1ac69f46a34538fd941255ee1dfb.a
/Users/trenskow/Library/Arduino15/packages/Intel/tools/arduino101load/2.0.1/arduino101load -c "-from=\"/var/folders/pz/kz0b20y15v7_hpvd7db9067c0000gn/T/arduino_build_193391/sketch_aug27a.ino.elf\"" "-to=\"/var/folders/pz/kz0b20y15v7_hpvd7db9067c0000gn/T/arduino_build_193391/../arduino101_sketch.debug.elf\""
signal: segmentation fault
/Applications/Arduino.app/Contents/Java/arduino-builder returned 255
Error compiling for board Arduino/Genuino 101.

```

per1234, 2023-01-29:
I received multiple reports of this bug from customers and can personally reproduce it. I'll share my findings here:

### Describe the problem

Compilation of any sketch for the Arduino 101 board fails with a `bsdthread_register` or `segmentation fault` error when using a computer with a recent version of macOS.

This is caused by the [**arduino101load**](https://github.com/facchinm/arduino101load) tool dependency having been built with an old version of [**Go**](https://go.dev/):

https://github.com/golang/go/wiki/MacOS12BSDThreadRegisterIssue

### To reproduce

#### Required equipment

- Mac computer with a recent version of macOS (Monterey or newer).

#### Steps

1. Select **File > New Sketch** from the Arduino IDE menus.
1. Select **Arduino IDE > Settings**
1. Check the box next to "**Show verbose output during: ☐ compilation**".
   **ⓘ** This step is not required. It is added so that the specific compilation command triggering the error can be seen in the output.
1. Select **Tools > Board > Intel Curie Boards > Arduino/Genuino 101** from the Arduino IDE menus.
1. Select **Sketch > Verify/Compile** from the Arduino IDE menus.

🐛 Compilation fails:

```text
/Users/per/Library/Arduino15/packages/Intel/tools/arduino101load/2.0.1/arduino101load -c "-from=\"/private/var/folders/sx/07hx_hp11nz50sf4ljwylj080000gn/T/arduino-sketch-87ED0F6F2BE4123CD63C6A26DDC030C6/sketch_jan29a.ino.elf\"" "-to=\"/private/var/folders/sx/07hx_hp11nz50sf4ljwylj080000gn/T/arduino-sketch-87ED0F6F2BE4123CD63C6A26DDC030C6/../arduino101_sketch.debug.elf\""
signal: segmentation fault

Compilation error: signal: segmentation fault
```

### Expected behavior

Compilation of valid sketches for Arduino 101 board is possible users of modern macOS machines.

### Additional context

The above demo was on an Apple Silicon-based Mac machine running the [**Rosetta 2** interpreter](https://en.wikipedia.org/wiki/Rosetta_(software)#Rosetta_2) because that is the only machine I have access to, but the same problem also occur on an x86 Mac, as reported by users of the Arduino 101 board:

- https://forum.arduino.cc/t/compilation-error-exit-status-2/1079262
- https://forum.arduino.cc/t/arduino-ide-is-not-working-with-intel-curie-board-on-macos/1046077

On the x86 machine, the error message is a bit more verbose:

```text
/Users/x-user/Library/Arduino15/packages/Intel/tools/arduino101load/2.0.1/arduino101load -c "-from=\"/private/var/folders/dl/cdjfr_wx1lv48vd5mzr4klj80000gp/T/arduino-sketch-5CEDF04D633012E523AC2E5C948FE18F/Blink.ino.elf\"" "-to=\"/private/var/folders/dl/cdjfr_wx1lv48vd5mzr4klj80000gp/T/arduino-sketch-5CEDF04D633012E523AC2E5C948FE18F/../arduino101_sketch.debug.elf\""

runtime stack:
runtime.throw(0x1122f8, 0x21)
	/ssd/go/src/runtime/panic.go:566 +0x95 fp=0x7ff7bfeff5c0 sp=0x7ff7bfeff5a0
runtime.goenvs()
	/ssd/go/src/runtime/os_darwin.go:88 +0xa0 fp=0x7ff7bfeff5f0 sp=0x7ff7bfeff5c0
runtime.schedinit()
	/ssd/go/src/runtime/proc.go:450 +0x9c fp=0x7ff7bfeff630 sp=0x7ff7bfeff5f0
runtime.rt0_go(0x7ff7bfeff660, 0x4, 0x7ff7bfeff660, 0x1000, 0x4, 0x7ff7bfeff810, 0x7ff7bfeff86c, 0x7ff7bfeff86f, 0x7ff7bfeff8ee, 0x0, ...)
	/ssd/go/src/runtime/asm_amd64.s:145 +0x14f fp=0x7ff7bfeff638 sp=0x7ff7bfeff630

exit status 2

Compilation error: exit status 2}
```

---

Related: https://github.com/facchinm/arduino101load/issues/3

per1234, 2023-02-17:
Good news everyone! This bug has been fixed in the latest 2.0.6 release of the "**Intel Curie Boards**" platform. The error should no longer occur after you use the Arduino IDE Boards Manager to install or update to version 2.0.6.

---

A note about the "**Intel Curie Boards**" 2.0.6 release: this release is identical to [2.0.5](https://github.com/arduino/ArduinoCore-arc32/releases/tag/2.0.5) except for updating the `arduino:arduino101load` tool dependency to version [2.0.3](https://github.com/arduino/arduino101load/releases/tag/2.3.0). Since it was done by a team not involved in the maintenance of this repository and only required changes to the package index file hosted in another repository, a `2.0.6` tag hasn't been pushed to this repo yet.

facchinm, 2023-02-20:
@per1234 tag 2.0.6 pushed, closing this issue :slightly_smiling_face:

closed 2023-02-20 by facchinm